butterstick-fpga/butterstick-hardware #4: Ethernet MDIO line needs pullup

gregdavill, 2019-12-08:
Add a 1K-4k7 pull-up to the MDIO line to the PHY.

Nightsd01, 2019-12-14:
Hey Greg, I am so impressed by this project but I have a quick question for you and couldn’t find a better place to ask. I’m a bit of a n00b with PCB design and I was wondering why some of the traces on your board seem to be so “squiggly” (for lack of a better word)?

Does it have something to do with avoiding signal interference? Or are you using it to make sure traces are of equal length even if they go to different places of the PCB?

[image]

gregdavill, 2019-12-15:
They're meandering traces. They are used to ensure signals transmitted from one chip arrive at the same time on the receiving chip.

closed 2020-11-01 by gregdavill