butterstick-fpga/butterstick-hardware #12: LED G & B control signals are crossed.

TomKeddie, 2021-12-12:
In this schematic excerpt you can see LEDG is driving CATB and LEDB is driving CATG.  Is easily fixed in the platform definitions but the schematic labels could be updated to swap LEDG and LEDB for clarity.

[image]

nandland, 2022-06-06:
I found this issue as well... scratching my head wondering why the Blue LED is turning on when I specifically asked for green. :)

gregdavill, 2022-06-07:
Thanks for picking this up Tom.

The easiest fix is to swap the signals from the FPGA to this page. Q2/Q3 will still control Blue/Green respectively. But now the signals are correctly named from the FPGA.

commit fac6a834 (2022-06-07): Fix RGB G/B signals

- Corrects a mistake where blue and green signals were swapped
- Regenerate schematic

Closes #12
closed 2022-06-07 by gregdavill