butterstick-fpga/butterstick-hardware #13: Are you planning on adding a license any time soon?

francis2tm, 2022-03-03:
Hello,
Amazing work on this board, are you planning on adding a license? Like CERN OHL v2? https://ohwr.org/project/cernohl/blob/master/licence_texts/cern_ohl_p_v2.txt

Thanks!

gregdavill, 2022-03-03:
The hardware README's have a mention of a license: https://github.com/butterstick-fpga/butterstick-hardware/tree/main/hardware/ButterStick_r1.0#licence

I'll make this more clear.

mithro, 2022-03-03:
I recommend adding a LICENSE or COPYING file, it is best practice and GitHub will then display license information on the repo too.

mithro, 2022-03-03:
FYI - https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-license-to-a-repository

gregdavill, 2022-03-04:
Added in 3a45b19e4130355be5552e34cc4e072c1e7afb3d

closed 2022-03-04 by gregdavill