butterstick-fpga/butterstick-hardware #17: Assembled PCBs link don't work

LaneaLucy, 2023-07-21:
When I open the link, I only get the homepage and when I search for butter stick on it, I only find the SYZYGY Breakout board for Butterstick

gregdavill, 2023-07-27:
I'll check but this probably means that the first batch of boards has entirely sold out.

LaneaLucy, 2023-07-28:
Is there any other option to get a preassembled one, maybe even with other ram chips? Found the MT41K512M16VRP-107 which should be pin compatible, except that one pin, that is NC is a extra address pin, but the schematic show, that this pin is already connected. And I would really like to get one with even more ram

gregdavill, 2023-08-26:
Closing because the link on the main page now works again. (while stock is available)

I'm able to do rework for different RAM alterations. Just drop me an email, it's listed on my profile :+1:

closed 2023-08-26 by gregdavill