butterstick-fpga/butterstick-hardware #18: VID seems... wrong?

Goof512, 2023-09-03:
Hi,

I'm working on getting the linux-on-litex-riscv project to build onto butterstick, and I noticed the ID from `lsusb` is showing up as "1209:5af1", whereas documentation seems to suggest it should be "0403:6014"

Is this a manufacturing revision that wasn't documented? or is this a *me* problem that I just didn't realize I'd walked into?
DFU upload using the tutorial doesn't show any errors, and flashes the board correctly, which confuses me deeply.

I'm kinda new to FPGA toolchains, so I'm not quite sure how to proceed.

Environment:

Mac x86 on Monterey 12.6
open-ocd v0.10.0 from xPack
yosys etc. from oss-cad-suite

Goof512, 2023-09-04:
After more research, it appears there is no option to use DFU to

closed 2023-09-04 by Goof512

gregdavill, 2023-09-04:
Do you have a link to any documentation that lists `0403:6014`? As that's the PID/VID pair for FTDI 2232H, not the DFU interface.

Goof512, 2023-09-04:
that was entirely the problem, yes. I closed this because it was a stupid mistake - I hadn't realized the OpenOCD configuration was designed against a separate programmer.